MotorDynamicsLab/Leviathan #2: Marlin source

thisiskeithb, 2023-11-24:
Hello!

Keith here from the [Marlin Firmware team](https://github.com/MarlinFirmware/Marlin).

Jason was kind enough to ship over one of these Leviathan boards to add Marlin support, but starting from scratch is fairly involved and I was hoping to collect some more information/files from a board developer. I can create the required pins file based on the [Klipper configs hosted within this repo](https://github.com/MotorDynamicsLab/Leviathan/tree/master/Firmware) and labeling on the board, but I don't have the PlatformIO variant files or build environment details which are crucial to getting it up and running.

We have a few boards with an `STM32F446ZET6` MCU, namely the BTT Octopus Pro V1.0, so I planned to copy those variant files over as a starting point just to see if I can get it going, but I doubt it'll be that easy 😄.

As an example of what I'm after, [here's a PR I put into Marlin](https://github.com/MarlinFirmware/Marlin/pull/23008/files) adding support for a board. These files in particular would be helpful:
- `buildroot/share/PlatformIO/boards/<board>.json`
- `buildroot/share/PlatformIO/variants/<variant>/*`
- Variant details in `ini/stm32f4.ini`.

If you have a working Marlin source, I can get that prepped & ported upstream, but any info would be greatly appreciated!

JNP-1, 2023-11-24:
> Hello!
>
> Keith here from the [Marlin Firmware team](https://github.com/MarlinFirmware/Marlin).
>
> Jason was kind enough to ship over one of these Leviathan boards to add Marlin support, but starting [...]

Nice to have you here. I'll have a look at it. I have to set up PlatformIO first.
To be honest, I haven't done anything with Marlin for a long time.
But it shouldn't be a problem.

thisiskeithb, 2023-11-25:
I'm going to close this issue since @meteyou submitted a PR and we're working through the required changes.

Thanks!

closed 2023-11-25 by thisiskeithb