MotorDynamicsLab/Leviathan #3: Manual Diagram Out of Date (Page 13)

ChrisWhite1985, 2023-12-08:
Please can you update the manual on page 13 to show the latest PCB layout.

Also can you share the PCB Socket Types so we can acquire the correct plugs and include how best to connect the board to RPI?

JNP-1, 2023-12-11:
> Please can you update the manual on page 13 to show the latest PCB layout.
> 
> Also can you share the PCB Socket Types so we can acquire the correct plugs and include how best to connect the board to RPI?

Page 13 already shows the current layout (V1.2).
All sockets are of the JST-XH type.
A ready-made cable with adapter board is already included for the Pi power supply.
It is also described here with pictures in the Raspberry Pi Mounting section.
https://www.ldomotion.com/p/guide/VORON-Leviathan-V12

ChrisWhite1985, 2023-12-11:
Ah I can see that Page 13 shows a Raspberry Pi installed on the Leviathan which is where my confusion started. This should be made clear i.e. "Leviathan V1.2 with Raspberry Pi 4 Model B installed". As RPI is not included with the Leviathan board and requires the user to install it seperately?

JNP-1, 2023-12-11:
> Ah I can see that Page 13 shows a Raspberry Pi installed on the Leviathan which is where my confusion started. This should be made clear i.e. "Leviathan V1.2 with Raspberry Pi 4 Model B installed". As RPI is not included with the Leviathan board and requires the user to install it seperately?

To be honest, page 13 is all about the dimensions.
Mounting material is included with the board, the RPi must be installed by the user.

ChrisWhite1985, 2023-12-11:
Sure, having seen the above link with information on how the RPI mounts its much clearer. What type of plug is the CANBUS port as well please, I think this is also not indicated?

closed 2023-12-11 by ChrisWhite1985