MotorDynamicsLab/Leviathan #4: no 12v supply for fan headers

TheRealDeathsneeze, 2023-12-27:
The general selection for fans in 12v is much better than with 24v.  Other contemporary boards support 12v  fans. Leviathan should too.

kingnoob1377, 2023-12-28:
Yes, 12V fans are much more common, so not having any 12V port is such a bummer.

JNP-1, 2024-01-02:
Is this about a specific use case?

TheRealDeathsneeze, 2024-01-04:
yes. 12v fans that are not available in 24v or 5v variants. specifically 1U server fans.

JNP-1, 2024-01-05:
424J/2HP from ebmpast are 24V 1U server fans.
https://www.ebmpapst.com/de/en/products/axial-fans/p/424JH.html

TheRealDeathsneeze, 2024-01-10:
I wasnt aware of that option, however it's fairly low flow and somewhat low temperature rating. 

a 12v option on the board would open up the selection for fans greatly. Would love to see it on a revision of the board

JNP-1, 2024-01-10:
May I ask what you want to do with a high flow fan on a printer?

CallumRD1, 2024-02-02:
I also miss the 12V fan option on the Leviathan. I have a stash of 12V Noctua electronics fans that I use on printers and the best 4010 part cooling fans for Xol or Blackbird/Archetype toolheads are 12V.

JNP-1, 2024-02-07:
Noctua now also has 24V. Another option would be to use a 24V fan and throttle it via PWM.

CallumRD1, 2024-02-07:
While many 24V fans do exist, that doesn’t remove the utility of having a 12V option. I’m currently forced to run 12V buck converters to get my 12V part cooling and electronics fans to run on Leviathan. It was so much more convenient in my octopus where I could just select 12V output. And these fans don’t exist in 24V configurations, so even if I wanted to replace ~$100 in fans for native Leviathan support, I couldn’t. It’s really obnoxious that such a simple and useful feature was omitted from this board.

JNP-1, 2024-02-07:
The decision did not come from me.

TheRealDeathsneeze, 2024-02-07:
> Adding an option to use 12V could be very simple, assuming 12V is available already and the real-estate on the PCB is available as well. Mellow does it with the Fly Super8 too.

Not to mention since the board has 5160s built in, there needs to be a 12v source on the board anyways.

JNP-1, 2024-02-07:
If someone short-circuits these 12V because a fan is defective, the TMC5160 would also be destroyed during operation. This is the reason why these 12V are not available for other things.

CallumRD1, 2024-02-08:
> The decision did not come from me.

We're not trying to direct any criticism towards you directly, just having a conversation about a feature we feel this product is missing.

I, for one, find this enough of an issue that I'm directing a few people I know away from Leviathan because I know that they require 12V fan support and the hassle of figuring out how to deal with alternative 12V fan control is beyond them. 

If having the fans on the same 12V rail as the 5160s is an issue then add a second 12V rail dedicated to fans. I'm not saying it's necessarily easy to implement, but I think it's such an important feature that it is with the effort. I think the Leviathan would be a much more compelling product with it. The Leviathan is so close to being an ideal board for most standard printer configurations that it's a shame to see it undercut by a decision like this.

JNP-1, 2024-02-08:
I will take it into account in the next version.