MotorDynamicsLab/Leviathan #5: Unable to update firmware after updating klipper - mainsail

midicdj1000, 2023-12-30:
I just flashed the Leviathan with klipper 11 and then told mainsail to update everything. Now I can't flash new firmware it says it can't find the usb port.

I press sw1 twice and the boot loader led flashed. 

Run the command to check it is showing 

pi@mainsailos:~ $ ls /dev/serial/by-id/*

/dev/serial/by-id/usb-katapult_stm32f446xx_1B0047000451313133353932-if00

I run the flash command
~/klippy-env/bin/python ~/katapult/scripts/flashtool.py -d /dev/ttyACM0 -f out/klipper.bin

and it gives this error

pi@mainsailos:~/klipper $ ~/klippy-env/bin/python ~/katapult/scripts/flashtool.py -d /dev/ttyACM0 -f out/klipper.bin
ERROR:root:Flash Error
Traceback (most recent call last):
  File "/home/pi/klippy-env/lib/python3.9/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/katapult/scripts/flashtool.py", line 543, in run
    serial_dev.open()
  File "/home/pi/klippy-env/lib/python3.9/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/katapult/scripts/flashtool.py", line 629, in main
    loop.run_until_complete(sock.run(args.device, args.baud, fpath))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/katapult/scripts/flashtool.py", line 545, in run
    raise FlashCanError("Unable to open serial port: %s" % (e,))
FlashCanError: Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
pi@mainsailos:~/klipper $

midicdj1000, 2024-01-02:
With some help from the klipper discord this command worked

~/klippy-env/bin/python ~/katapult/scripts/flashtool.py -d /dev/serial/by-id/usb-katapult_stm32f446xx_1B0047000451313133353932-if00 -f out/klipper.bin

it needed the full dev device instead of the ttyACM0

camerony, 2024-01-02:
Can I get some more information?  What versions of the RPi are you using?  Also the OS version, Bookworm or Bullseye?

midicdj1000, 2024-01-03:
[image]

PI 3B+

How do I check for /dev/ttyACM0 ?

camerony, 2024-01-03:
`ls /dev/tty*`

midicdj1000, 2024-01-03:
[image]

midicdj1000, 2024-01-03:
how do you know what it's allocated to ?
something must have changed between the first flashing and all the upgrades.

midicdj1000, 2024-01-03:
I forgot to show cmdline
[image]

camerony, 2024-01-03:
That is your USB port, not sure what happened.  But it should work with /dev/ttyACM0, if the Leviathan is the only device plugged in to the Rpi.

midicdj1000, 2024-01-03:
It did work the first time before all the updates

camerony, 2024-01-03:
Does it work now though?  I'll haven't been able to reproduce your issue, since /dev/ttyACM0 exists.

midicdj1000, 2024-01-03:
and it does not program
[image]

midicdj1000, 2024-01-03:
if I change to ttyACM1 it programs ok
[image]

camerony, 2024-01-03:
Ok, thanks.  Not sure why you're not getting /dev/ttyACM0 in bootloader mode.  But at least we confirmed it does work, just that it got reassigned as /dev/ttyACM1.     I'll see if can reproduce that.

midicdj1000, 2024-01-03:
it would be a lot easier to use the /dev/serial/by-id/usb-katapult_stm32f446xx_1B0047000451313133353932-if00 with the unique id, you ask to check this is present so why not use it and save any potential problems with the ACM0

camerony, 2024-01-03:
true, I'll add that is a second option.  Thanks for troubleshooting.

midicdj1000, 2024-01-03:
Can you explain the USB, CAN and UART options at the start.
If I am using UART to talk to the PI will this stop me using CAN to connect to a toolhead board ?

I setup with USB to talk to the PI and want to use  CAN connect to a BTT board once I have everthing else setup.

camerony, 2024-01-03:
I suggest you join https://discord.com/channels/692982154577313814/1180911286142369842   for that question.

camerony, 2024-01-03:
[image]
 
added addition command option