MotorDynamicsLab/Leviathan #8: other users

Magneticrob1, 2024-01-14:
are there any other users of this board on here, i have put bootloader and firmware on but cant connect to any external can modules. is there a original bootloader file available anywhere

midicdj1000, 2024-01-16:
I am trying to connect to the EBB36 V1.2 and I think I have the same problem, it only has one can bus ID when it's connected.
I also don't know how to setup the MCU for the can bridge.

midicdj1000, 2024-01-17:
I flashed my EBB board with the stm programmer and I have it working.

Mattang-Tim, 2024-02-05:
I was able to get mine flashed/connected up using the CB1 eMMC with the Pi4B board as my Pi.

JNP-1, 2024-02-07:
> are there any other users of this board on here, i have put bootloader and firmware on but cant connect to any external can modules. is there a original bootloader file available anywhere

Have you also activated the CAN interface under Linux?