MotorDynamicsLab/Leviathan #9: usage of tach wire with 5v delta fans results in fan shutting off

TheRealDeathsneeze, 2024-02-02:
When the tachometer pin on a 5 volt 3 wire Delta axial fan is used with any of the fan ports, it results in the fan shutting off instead of spinning and no tachometer output is given. 

I had initially thought that there was an error with the listing for these fans, as the tachometer wire is blue instead of the typical yellow. However, when plugged into another known working device, the fan behaves as expected.

TheRealDeathsneeze, 2024-02-02:
Of note is that a 4020 12v Delta 4 wire fan(model THA0412AD) I have on hand does function as expected. The tested fan not working as expected is ASB02505SHA-AY6B. See attached datasheets for both fans. 
[ASB02505SHA-AY6B.pdf](https://github.com/MotorDynamicsLab/Leviathan/files/14144702/ASB02505SHA-AY6B.pdf)
[THA0412AD.pdf](https://github.com/MotorDynamicsLab/Leviathan/files/14144703/THA0412AD.pdf)

JNP-1, 2024-02-03:
I looked at both data sheets and could not see any difference in the tach pin.
Even if you connect the tach pin to GND, the fan must work.
On Leviathan, the tach pin is also "decoupled" and protected with a diode.
This must have another cause.

TheRealDeathsneeze, 2024-02-07:
the working fan is 12v. As best as I can tell the tach is using 15v logic. 
the non working fan is 5v. I'm guessing using 5v logic? Either way, it functions as expected when plugged into an endstop port on an ebb36. 

See attached reference images. 

[image]
[image]

TheRealDeathsneeze, 2024-02-07:
[image]

[image]

JNP-1, 2024-02-07:
The tach output of a fan is an open collector output.
This means that the level depends on the signal from the pull-up resistor (to which voltage it is connected) and not on the operating voltage of the fan.
In the case of Leviathan, this is a 3.3V signal.

TheRealDeathsneeze, 2024-02-07:
oh so in this case, perhaps the reason it does not function is expected is due to the 3.3v pullup not being a 5v pullup? 

This is a fair bit outside of the scope of my knowledge, is there a way I could easily test to see if this is the root of the problem?

JNP-1, 2024-02-07:
The MCU only needs a 3.3V signal to recognize the tacho.
So I don't understand why it shouldn't work with a 5V fan.