MotorDynamicsLab/Leviathan #12: Limited capacity LED driver for strip lighting?

jetcat-bob, 2024-02-06:
My LDO Voron 2.4 kit came with 2 strip LEDs for the chamber light. Their combined current requirement is 580ma @ 24v. The Leviathan has a limit of 350ma (LM3407MY). Is there a suggestion for a replacement LED strip that would be compatible with half the current of the supplied strip? It would be nice to have the same brightness. Maybe a more capable driver in the future would be nice......
[image]

JNP-1, 2024-02-06:
Perhaps the LED strip is too long. In my printer, I have a total of 36 LEDs (2x18) on the port.
Was the V2.4 an LDO kit?

jetcat-bob, 2024-02-06:
Yes it's a Voron 2.4 LDO kit. Two strips with 18 LEDs each. Each strip
pulls 280ma.

I was considering using a fan port on the Leviathan but I did find a lower
power LED strip that would work. 170ma per strip of 18. Not as bright
however.

I like the idea of having a constant current source for the lighting but
maybe on V2 consider a higher output device ;).

Bob Wilcox

JNP-1, 2024-02-07:
Then we both have the same LED strips, my printer is also an LDO kit.
With a constant current source, the output voltage is regulated to achieve the 350mA.
In my case I get 22.5V, the two LED strips are connected in parallel (with the included little PCB in the Kit).
Is it the same for you?

jetcat-bob, 2024-02-07:
Measuring with my DVM and not the iffy power supply, I get 20.8V @ 353ma.
Full brightness @ 24V is 510 ma. There is a difference but I can live with
it. Compared to my other Voron 2.4 that has 3 strips of 18 LEDs (24V @
700ma) on an Octopus fan port, it's pretty dim.

JNP-1, 2024-02-07:
It is clear that the current increases slightly at 24V. At 28V they are even brighter ;-)

ccatlett1984, 2024-02-08:
Just run positive and negative to the PSU, only the "signal" wire needs to be on the pin. if 3 wire.
Or use the spare heater output for them.

jetcat-bob, 2024-02-08:
I don't have a spare fan output but there is a spare heater. I'll look into
that possibility. Thanks for the suggestion!

Bob Wilcox

CallumRD1, 2024-02-08:
I’m also surprised at how low of a current the LED port is regulated to. My lights want to draw far more current than the header will output. It’s a very convenient port but isn’t usable for me due to the low brightness of LEDs connected to it. When I use a heater output with unregulated 24V the LEDs are far brighter. I’d have much preferred an unregulated LED header.

ccatlett1984, 2024-02-08:
You can use [output_pin] for that pin, and be able to control the
brightness.

JNP-1, 2024-02-08:
If you connect the LED strips in series, they should be brighter.

CallumRD1, 2024-02-08:
I can't. I'm using Daylight on a Stick and Matchlight on a Stick. They're designed to be run inline off of 24V.

ccatlett1984, 2024-02-09:
for those, you run the 24v directly to the psu, and the "data" pin to the
neopixel header on the board.

CallumRD1, 2024-02-09:
> for those, you run the 24v directly to the psu, and the "data" pin to the neopixel header on the board.
> […](#)

You have no idea what you’re talking about. These are not now pixels and have no control wire. They’re 24V white led strips with two pins: 24V and ground. They’re designed (like most white lite led strips for this kind of application) to take in straight, non current regulated 24V and daisy chain in parallel so you can run many of them inline.

jetcat-bob, 2024-02-09:
The LED strips are 2 wire. They are not addressable. For my case, I'll use
the the tool head heater output for the LEDs since the printer uses a
canbus driver board. The heater has plenty of current capability for making
the printer glow very brightly...

Bob Wilcox

ccatlett1984, 2024-02-09:
> > for those, you run the 24v directly to the psu, and the "data" pin to the neopixel header on the board.
> 
> You have no idea what you’re talking about. These are not neopixels and have no control wire. They’re 24V white led strips with two pins: 24V and ground. They’re designed (like most white light led strips for this kind of application) to take in straight, non current regulated 24V and daisy chain in parallel so you can run many of them inline.

I have a set that have the 3 pin connectors. Didn't realize they had 2 pin variants..
No need to be snobby about it.

Hecatron, 2025-06-22:
Stumbled across the issue myself just now, after the LED driver popped on the board
U15 / LM3407MY
I didn't think to check the current draw on the driver vs the led strip since both came in the LDO kit

[image]

JNP-1, 2025-06-23:
> Edit sorry I think this is a different problem The bust IC is the neopixel driver U15 / LM3407MY but I didn't have that plugged in I think what's happened in my case is the 5V rail might have developed a 78 ohm short to ground Which in turn might have spiked the 5V rail causing that IC (and possibly lots of others) to go boom, will check the 5V rail on my new one to see if it has a similar load
> 
> [image]

All voltage regulators on Leviathan are short-circuit protected.
I therefore do not believe that this is the cause.
U15 (74LVC1G17GW) is a 5V driver for the data line of the Neopixel.
Can you please open a new issue because it has nothing to do with the LED driver, thank you.