MotorDynamicsLab/Leviathan #16: Switching to DFU mode is no longer possible

MarkusNiewoehner, 2024-05-27:
For reasons unknown to me I can no longer switch to DFU mode. The Levithan is installed in a working printer and no other USB devices are connected

Here is the dmesg output after pressing the boot and reset buttons as described in the instructions:

biqu@Trident250:~$ dmesg -HW
[May27 12:31] usb 7-1: USB disconnect, device number 4
[  +0.000688] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.000228] gs_usb 7-1:1.0 can0: Couldn't shutdown device (err=-19)
[  +0.000281] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.003354] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.868004] usb 7-1: new full-speed USB device number 5 using xhci-hcd
[  +0.144559] usb 7-1: New USB device found, idVendor=1d50, idProduct=606f, bcdDevice= 0.00
[  +0.000008] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000004] usb 7-1: Product: stm32f446xx
[  +0.000003] usb 7-1: Manufacturer: Klipper
[  +0.000004] usb 7-1: SerialNumber: 350033000A51313133353932
[  +0.001290] gs_usb 7-1:1.0: Configuring for 1 interfaces
[  +0.000185] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.000323] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.146798] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.000802] IPv6: ADDRCONF(NETDEV_UP): can0: link is not ready
[  +0.000010] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.000046] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[  +0.000004] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing
[  +0.005909] [dhd] CFG80211-ERROR) wl_cfg80211_netdev_notifier_call : wdev null. Do nothing

JNP-1, 2024-06-04:
Do you have an STLink V2? You could use this to check whether the MCU is still accessible via the SWD interface.

MarkusNiewoehner, 2024-06-04:
Yes, but I've never used it before. I'll see if I can try that one day. I was now able to flash the Levi with the current klipper via catapult.

JNP-1, 2024-06-04:
Does the 3.3V LED light up when the board is powered?

MarkusNiewoehner, 2024-06-04:
In 24V operation or 5V USB? I can check tomorrow.

JNP-1, 2024-06-04:
In 24V operation.

MarkusNiewoehner, 2024-06-05:
> Does the 3.3V LED light up when the board is powered?

yes, she does